utexas-bwi/bwi #16: bwi_launch: roslaunch check failing

jack-oquin, 2015-03-12:


jack-oquin, 2015-03-12:
@piyushk: I am having trouble figuring out what to do with the various ``bwi_kr_execution`` references.

piyushk, 2015-03-12:
@jack-oquin I can take a look at this issue. I'm not exactly sure what you meant by your last commit though.

jack-oquin, 2015-03-12:
There are launch files in bwi_launch that reference objects in bwi_kr_execution. I added the missing dependency on that package, but the roslaunch check still fails. Not sure what's going on.

jack-oquin, 2015-03-12:
I only defined the dependency as an exec_depend, I think it needs either a build_depend or test_depend for the launch check to work. I'll try that.

jack-oquin, 2015-03-12:
That did not help. Somehow the reference is not getting resolved.

jack-oquin, 2015-03-12:
Maybe it's because bwi_kr_execution is failing to build.

See: utexas-bwi/bwi_common#15

jack-oquin, 2015-03-12:
@piyushk: the first commit fixed a bunch of references to segbot_apps instead of segbot_navigation, and so forth.

The formatting of some launch files was rather messed up. I found it necessary to eliminate hundreds of trailing blanks from many lines. Unfortunately, that obscured the actual changes. I should have split them into two commits. Sorry.

commit c10ef640 (2015-03-13): added missing dependencies. removed concert launch file as that code is not ready for primetime yet. closes #16.
closed 2015-03-13 by piyushk