utexas-bwi/bwi #19: Could not resolve rosdep key python-catkin-tools for distro heisenbug

piyushk, 2015-03-31:
I've split the relevant portion of #18 into this new separate issue.

piyushk, 2015-03-31:
@jack-oquin Could you check with Scott and figure out what the best way is to resolve this issue? I believe we decided that we were not going to drop the catkin-tools dependency.

jack-oquin, 2015-03-31:
Yes I will.

jack-oquin, 2015-03-31:
@cottsay: what is the recommended procedure for resolving things like this?

Is there a released version of ``python-catkin-tools`` for Fedora? If so, I'll just add the appropriate rosdep. If not, I am unsure how to proceed. We had to skip Heisenbug on our previous release.

cottsay, 2015-03-31:
There is. The package is called `python-catkin_tools`.

See: https://admin.fedoraproject.org/pkgdb/package/python-catkin_tools/

Thanks for bringing this up instead of skipping generation again!

--scott

jack-oquin, 2015-03-31:
Anskthay, @cottsay! 

I'll add the missing rosdep.

jack-oquin, 2015-04-02:
Re-released bwi verision 0.3.1 to update the release repository with Heisenbug support, but no new pull request.

Hopefully, the build farm will pick this up and provide Heisenbug binaries for this package.

jack-oquin, 2015-04-04:
@cottsay: the release package contains Heisenbug tags now. 

How can I verify that it is being packaged correctly?

cottsay, 2015-04-04:
We don't have any "devel" jobs or anything like that in Fedora right now, so its hard to test. I'll clone the release repo and give it a spin on the firm quick. First glance says everything looks okay though...

cottsay, 2015-04-04:
`x86` and `x86_64` Builds succeeded:
- http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-bwi-desktop_sourcerpm/
- http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-bwi-desktop-full_sourcerpm/
- http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-bwi-launch_sourcerpm/

ARM builds never kicked off because `libsegwayrmp` is failing on ARM [1].

You should be good to release `0.3.1-1` via `rosdistro` now. I'm trying to take care of some ARM bugs in the buildfarm, but I hope to sync building->shadow-fixed this weekend. Maybe even sync to public, we'll see...so in any case, it should hit the (un)stable repos soon.

Thanks!

[1] http://csc.mcs.sdsmt.edu/jenkins/job/ros-indigo-libsegwayrmp_binaryrpm_21_armhfp/3/consoleFull

cottsay, 2015-04-21:
@jack-oquin, do you plan to release `0.3.1-1` any time soon? `0.3.1-0` appears to be the newest in `rosdistro`, but `0.3.1-1` is the newest in `bwi-release`...

jack-oquin, 2015-04-22:
I didn't realize I was creating a new patch level, and thought the build farm would somehow pick up those changes automatically.

I will go ahead and release 0.3.1-1. Thanks for pointing out my error.

jack-oquin, 2015-04-23:
Patch released: ros/rosdistro#8113 and ros/rosdistro#8114

closed 2015-04-23 by jack-oquin