utexas-bwi/bwi #20: roberto: sonar board not working

jack-oquin, 2015-04-18:
It always returns "nothing found".

The one on flexo seems OK.

jack-oquin, 2015-04-18:
I can work on this next Wednesday, April 22nd.

maxsvetlik, 2015-04-21:
What is the sketch we currently use for arduinos on segbot_v2s? Multidevice.ino?

jack-oquin, 2015-04-21:
Yes

piyushk, 2015-07-21:
I've verified that re-uploading the sketch does nothing. Will attempt replacing arduino/sonars tomorrow.

piyushk, 2015-07-21:
@jack-oquin I've been testing for the last 2 hours, and I think we've got a bigger problem - I think the quality of the sonar sensors is the root cause here.

I scavenged every sonar sensor I could find in the lab, and tested across 3 arduinos, 2 sets of cables and our custom shield and all these old sensors. I got some readings in the middle, but after switching to a different set of cables and shield, those readings also went away.

I know these sonars had issues in the past when we used a different power supply than USB, but I suspect there are other issues. For the time being, let's only use the voltage output from the arduino.

For segbot v3, we'll be using these sensors instead: http://www.maxbotix.com/Ultrasonic_Sensors/MB7092.htm. At that point we'll just make 4 more boxes for segbot v2, and fix things accordingly. Until then, I won't incorporate the sonars in the navigation code.

jack-oquin, 2015-07-21:
OK.

They did not seem very reliable to me, either.

jack-oquin, 2015-07-22:
Should we close this issue, or leave it open for testing new devices?

piyushk, 2015-07-22:
I'll create a separate issue for new sonar devices once they arrive.

closed 2015-07-22 by piyushk