utexas-bwi/bwi #21: Indigo: intermittent localization problems

jack-oquin, 2015-05-20:
This was observed recently on **flexo**, the only Trusty/Indigo system currently running. 

To reproduce:

* run low-level robot navigation: ``roslaunch bwi_launch segbot_v2.launch``
* make sure the particle filter spread is visible in ``rviz``
* assign various x, y navigation goals via ``rviz``

After a while, the particle filter spreads out all over the place. 

So far, the root cause is unclear. Maybe the navigation stack works differently in Indigo. Maybe the **flexo** base is periodically generating bogus odometry readings. Maybe it's something else entirely.

Shiqi installed Trusty and Indigo on the **clamps** laptop. I plan to try that on one of the other segbots.

cc: @shiqizhang6, @mleonetti, @maxsvetlik, @jsinapov

piyushk, 2015-05-26:
@shiqizhang6 and I tracked down the bogus odometry readings to https://github.com/segwayrmp/segway_rmp/issues/25. We get bogus odometry readings only when we run the roslaunch command **before** turning the segway base on. If we turn the segway base on first, it should not happen. A fix for the issue is incoming, but tests should make sure that the segway base on first.

I also talked with @jsinapov a while back, and the localization spread seems a bit much. We'll have to designate someone to test out those parameters soon.

jack-oquin, 2015-05-26:
I feel sure that the base was already on before running these tests. If I forget: I habitually kill the original roslaunch; turn on the base; then run everything again.

jack-oquin, 2015-06-10:
There are many more checksum errors then usual reported on the segway1 base ( **flexo**). It seems that some bogus messages may be getting through without the checksum algorithm detecting them. That could explain occasional large odometry jumps, observed by @piyushk.

piyushk, 2015-06-10:
As per William's suggestion in https://github.com/segwayrmp/segway_rmp/issues/26, it's probably best to add an odom filter node to catch these kinds of errors and suppress them.

jack-oquin, 2015-08-01:
This problem has not recurred since the failing Segway base was retired.

Closing as a hardware problem.

closed 2015-08-01 by jack-oquin