utexas-bwi/bwi #22: Segbot batteries: need endurance tests

jack-oquin, 2015-05-20:
Some of the batteries are getting old and probably need replacement.

On **roberto**, we plan to track the ``/battery0`` topic output to measure the voltage drop over time. The older segbots do not have a voltage meter built in, so we can only measure how long the robot runs until it starts beeping.

cc: @maxsvetlik, @mleonetti, @jsinapov, @shiqizhang6

shiqizhang6, 2015-05-20:
We've collected a ROS bag file that recorded the battery voltage over 3 hours, more or less. The robot started with a full battery and was turned off when it started beeping. 
http://www.cs.utexas.edu/~szhang/tmp/

jack-oquin, 2015-05-20:
The data above were collected with **bender**. Since it ran over 3 hours, we believe that battery is working well enough.

shiqizhang6, 2015-05-21:
Jack, I think the voltage data was collected with *Roberto* - the only one that has the voltage meter (at least the laptop on it now is Roberto). I've tested Bender's battery today, and it supported a 2:45 hours run. It did not beep when terminated because I had to leave then.

shiqizhang6, 2015-05-22:
Marvin's battery supported a run of 1:02 hours this afternoon. What was not so good is that the converter and the Segway base started beeping at the same time.

jack-oquin, 2015-05-24:
Quite right, it was **roberto**, which is on a segbot2 with a voltage meter.

Sounds like **marvin** needs a new battery.

closed 2016-09-15 by jack-oquin