utexas-bwi/bwi #23: LDAP: login shell is /bin/sh, should be /bin/bash

jack-oquin, 2015-07-09:
I did some config updates that were supposed to fix this, but bash (and zsh) are not in the list of possible shells for LDAP configuration.

cc: @piyushk

jack-oquin, 2015-07-09:
I was able to update the shell via phpLDAPadmin, by adding a ``loginShell`` field set to ``/bin/bash``.

It did not work when initially creating the user ID.

jack-oquin, 2015-07-09:
I fixed the three actual users already defined: joq, piyushk and bwi.

I'll leave this open until we can create a new user with the proper shell.

jack-oquin, 2015-07-15:
The currently-recommended method of creating a new user is to copy from and existing one. Among other advantages, that approach bypasses this problem.

closed 2015-07-22 by jack-oquin

arctiqjacob, 2018-08-22:
This can be fixed by editing: /usr/share/phpldapadmin/templates/creation/posixAccount.xml and at line 75, changing `<value id="/bin/sh">Bash</value>` to `<value id="/bin/bash">Bash</value>`