utexas-bwi/bwi #24: LDAP: define user ID's for current BWI staff

jack-oquin, 2015-07-09:
The previoius user list retained from Hydro was:

```
mbroussard:x:10005:10002:Matthew Broussard,,,,:/nishome/mbroussard:/bin/bash
ocroomes:x:10008:10002:Oliver Croomes,,,:/nishome/ocroomes:/bin/bash
stkim:x:10022:10002:Thomas Kim,,,:/nishome/stkim:/bin/bash
fkyang:x:10053:10000:Fangkai Yang:/nishome/fkyang:/bin/bash
jthywiss:x:10058:10000:John Thywissen,,,:/nishome/jthywiss:/bin/bash
matish:x:10059:10000:Matteo Leonetti,,,:/nishome/matish:/bin/bash
jliang:x:10065:10000:Jason Liang,,,:/nishome/jliang:/bin/bash
gdc_touch:x:10068:10000:gdc_touch,,,:/nishome/gdc_touch:/bin/bash
szhang:x:10070:10000:Shiqi Zhang:/nishome/szhang:/bin/bash
stephenl:x:10071:10006:Stephen Langevin,,,:/nishome/stephenl:/bin/bash
shuuh:x:10078:10006:Susmita Hossain:/nishome/shuuh:/bin/bash
BrianW:x:10082:10006:Brian Wang,,,:/nishome/BrianW:/bin/bash
YuqianJ:x:10097:10006:Yuqian Jiang,,,:/nishome/YuqianJ:/bin/bash
JonathanC:x:10102:10006:Jonathan Christian,,,:/nishome/JonathanC:/bin/bash
msvetlik:x:10107:10006:Maxwell Svetlik,,,:/nishome/msvetlik:/bin/bash
jsinapov:x:10108:10000:Jivko Sinapov,,,:/nishome/jsinapov:/bin/bash
mkurian:x:10109:10006:Mathew Kurian,,,:/nishome/mkurian:/bin/bash
kdeholton:x:10110:10006:Kyle Deholton,,,:/nishome/kdeholton:/bin/bash
plankenau:x:10111:10006:Patricio Lankenau,,,:/nishome/plankenau:/bin/bash
jthomason:x:10112:10000:Jesse Thomason,,,:/nishome/jthomason:/bin/bash
aashokkaushik:x:10113:10000:Arati Ashok Kaushik,,,:/nishome/aashokkaushik:/bin/bash
pkhante:x:10114:10000:Priyanka Khante,,,:/nishome/pkhante:/bin/bash
rahmadzadeh:x:10116:10007:Reza Ahmadzadeh,,,:/nishome/rahmadzadeh:/bin/bash
sarah:x:10117:10002:Sarah Jee,,,:/nishome/sarah:/bin/bash
ggaudioso:x:10118:10000:Ginevra Gaudioso,,,:/nishome/ggaudioso:/bin/bash
jlim995:x:10119:10001:Jeremy Lim,,,:/nishome/jlim995:/bin/bash
max:x:10121:10000:Maxwell Svetlik,,,:/nishome/max:/bin/bash
carson:x:10122:10001:Carson Moore,,,:/nishome/carson:/bin/bash
```

We probably don't need them all any more. The priority should be the summer staff:

* [ ] max
* [ ] jsinapov
* [ ] YuqianJ
* [ ] szhang
* [ ] jthompson
* [ ] pkhante
* [ ] mleonetti (missing in the list above)

jack-oquin, 2015-07-09:
Yuquan should probably follow standard Unix and Linux practice and define an entirely lower-case user ID.

jack-oquin, 2015-07-09:
Since I don't know people's passwords and do not at want to define anything guessable, I will initially define them all to use the ``bwi`` password, and ask everyone to change it themselves.

But, see #25.

jack-oquin, 2015-07-09:
To migrate each user **foo**:

 * Create user **foo** with the same UID in phpLDAPadmin, with group **bwi** or **fri**. 
 * For system admin access, add **foo** to the **sudo** and **dialout** LDAP groups.
 * Move ``/nishome/foo`` to ``/home/users/foo``.
 * ``userdel foo``

piyushk, 2015-07-09:
@jack-oquin Every user will need access to the dialout group.

jack-oquin, 2015-07-09:
OK, noted above.

closed 2015-07-15 by jack-oquin