utexas-bwi/bwi #26: nibbler: fails to boot

jack-oquin, 2015-07-09:
This may have been caused by a faulty LDAP client configuration. I made one attempt to fix it, but no luck yet.

To get read/write access to the the root filesystem:
 * boot into recovery mode
 * select root shell option
 * gain write access to the filesystem: ``mount -o remount,rw /dev/sda1 /``

jack-oquin, 2015-07-15:
I reinstalled this machine today.

Now the graphical login screen is not coming up, but the machine can be reached via ssh.

jack-oquin, 2015-07-15:
I fixed the lightdm config, and updated the instructions. That made graphical login start working again.

@jsinapov was then able to login using his LDAP account.

Afterwards, we rebooted the machine, and it is no longer on the network, ping returns ``Destination Host Unreachable``.

jack-oquin, 2015-07-16:
Reinstall did not help. Once LDAP was running again, the system would no longer boot.

Hypothesis: there is some system service starting before networking that needs user authorization, and it hangs due to PAM checking with LDAP first.

jack-oquin, 2015-07-16:
https://bugs.launchpad.net/ubuntu/+source/libnss-ldap/+bug/1024475

jack-oquin, 2015-07-16:
I think the links above point to the problem. Several work-arounds are suggested. Some look better than others. A fresh install will only work around this problem until we configure LDAP again, so we need to get to the bottom of this issue.

@piyushk: I have no time to come in today and fix this, but can probably come Friday. If you have time to work on it this afternoon, I recommend the following experiment, to demonstrate whether those links describe the root cause:

1. Boot into recovery mode.
2. Select the root shell option.
3. Mount the root filesystem read-write: ``mount -o remount,rw /dev/sda1 /``
4. Edit ``/etc/nsswitch.conf``, removing ``ldap`` from the three lines for passwd, group and shadow, so they just say ``compat``.
5. Resume booting.

If we have the problem I think we do, the boot should now succeed, but no one will be able to login via LDAP. At that point we can experiment with the various suggested alternatives.

piyushk, 2015-07-17:
Thanks for finding that bug report @jack-oquin! I couldn't get to it yesterday. I've got some errands to run today. I will be in for a few hours this afternoon and work on this problem, or I will work on it on the weekend. I'll keep you posted.

jack-oquin, 2015-07-17:
I'll plan to come to campus after lunch today.

jack-oquin, 2015-07-17:
I tried the experiment (above) on ``kif``, which was exhibiting the same failure as ``nibbler``.

Changing ``/etc/nsswitch.conf`` as in step 4, did bypass the problem, although step 5 failed to resume booting, and I had to reboot via power-on.

jack-oquin, 2015-07-17:
Tried several of the suggested options, no luck yet.

Trying ``libnss-ldapd`` in place of ``libnss-ldap``.

jack-oquin, 2015-07-17:
That seems to work.

closed 2015-07-17 by jack-oquin