utexas-bwi/bwi #27: LDAP: install and configure client software on lab machines

jack-oquin, 2015-07-09:
Status:
 * [x] hypnotoad
 * [ ] zapp
 * [ ] nibbler (broken, see #26)
 * [ ] kif

Are there any others to do?

piyushk, 2015-07-09:
I've added wernstrom, which likely needs a 14.04 installation as well.

jack-oquin, 2015-07-09:
Thanks. I couldn't remember that name. :smile:

jack-oquin, 2015-07-14:
I think we should add LDAP client software to the robots, too, but without automount.

piyushk, 2015-07-14:
+1, though we'll need to make sure that the robots are connected to the wireless prior to login. I think Network Manager in 14.04 supports this. 

Just in case the wireless stops working, we should make sure we can login locally to update network manager settings with new wireless user/pass.

jack-oquin, 2015-07-14:
Good point. I guess we should define a bwilab local account on every machine.

jack-oquin, 2015-07-16:
There are problems, see: #26

piyushk, 2015-07-22:
Unfortunately, it seems that network manager stores the password in the keyring, so it's not retrieved prior to login. https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/1158432 

I'm pretty sure that connecting to wifi worked on our Precise machines at the login screen. I'll continue investigating.

piyushk, 2015-07-23:
Before testing the /etc/network/interfaces route, I'm going to see if wicd solves the problem.

piyushk, 2015-07-23:
wicd requires all users to be a part of the netdev group by default. I'm gonna create a group locally and add bwilab to it. If wicd works, we'll remove the local group and add it to ldap.

jack-oquin, 2015-07-23:
Wicd looks like a good idea. Hope it works.

piyushk, 2015-07-23:
There are some issues with wicd as well. The wicd tray icon doesn't show up properly. (http://askubuntu.com/questions/580964/wicd-system-tray-icon-doesnt-show-up) and (https://bugs.launchpad.net/wicd/+bug/761326). Although wicd connected once prior to login, it failed a second time. Since there is no tray icon, it's hard to tell when wicd is connected or not. 

There are some workarounds, but they are likely nasty. I'm gonna fall back on the `/etc/network/interfaces` and `wpa_supplicant` route.

piyushk, 2015-07-23:
I've gotten LDAP to work correctly via wpa_supplicant and /etc/network/interfaces.

I'm documenting the process on the wiki right now on how to setup a robot.

piyushk, 2015-07-23:
bender and clamps have been updated to ldap using above mentioned. marvin is the arm robot, and roberto is still on hydro/precise for matteo's experiments.

I think the next thing to test is ldap automount. I'll do that tomorrow morning.

piyushk, 2015-07-24:
I'm gonna work on upgrading zapp, and then work on automount.

piyushk, 2015-07-24:
automount worked pretty easily, though the method of creating users and setting up machines needs to be updated. I've setup an nfs server on nixons-head again, and autofs-ldap automount directories.

I've also verified that pam_mkhomedir still works with the updated setup, which will be required on the robots. 

I'm updating instructions on the wiki right now, and will hard commit all the changes tomorrow morning since I do not want to commit them remotely.

piyushk, 2015-07-24:
Wiki instructions updated.

piyushk, 2015-07-25:
Darn it! The autofs client does automatically start correctly at first boot. I'm gonna investigate this issue tomorrow.

piyushk, 2015-07-25:
There are two problems (Both covered in this thread - http://ubuntuforums.org/showthread.php?t=2242635):
 1. autofs starts before network manager gets an IP address, and fails. Proposed solutions:
  - Use a cron job or a post connection trigger to restart autofs once boot is complete.
  - Use ifupdown and /etc/network/interfaces instead of network manager. *I will use this solution since it's less hackish and ethernet info doesn't need to be changed at all.*
 2. autofs mounts everyone's home directories instead of just the user who's logged in. 
  - I'm still working on a solution for this one.

jack-oquin, 2015-07-25:
1. ``/etc/network/interfaces`` seems fine to me, as long as it's documented in agonizing detail.

2. While it would be better only to mount the each user at login, I think we could live with mounting them all, as long as the directory permissions are set correctly.

piyushk, 2015-07-25:
@jack-oquin: I agree on both points. After I fix the first problem, I'll run the following test:
 1. Make sure wernstrom is already booted up
 2. Create a new user on nixons-head and create a home directory for user.
 3. Check on wernstrom before and after logging in to see if the home directory is mounted appropriately.

If the above test passes with the current setup, I'm fine letting problem 2 be. The directory permissions are setup correctly.

piyushk, 2015-07-26:
I've been working remotely to further improve machines, and have just realized that users 'bwi', 'wxie' and 'ggaudioso' exist both on the local system and ldap. I'm a bit afraid of implementing fixes remotely for user 'bwi', so I will investigate this problem in person tomorrow afternoon.

jack-oquin, 2015-07-27:
When I created those accounts, I did not realize that local versions already existed, and I forgot to check.

To correct my error:
 * change the UID in LDAP to the original value
 * delete the local user definition
 * move the ``/nishome/username`` directory to ``/home/users``.

piyushk, 2015-07-27:
Already done +1 :)

piyushk, 2015-07-28:
I've made the following changes today:
 * Installed nvidia drivers on all machines. nouveau was causing the machines to crash.
 * Disabled attempts by the clients to use authentication when contacting the LDAP server. This was slowing down LDAP requests and NFS automounts.
 * Update all desktops (including zapp) to use automount and LDAP.

Only the two robots clamps and roberto are left, which are both being used for experiments.

jack-oquin, 2015-07-28:
Is the wiki updated to reflect these changes? 

I can't tell, because it's "experiencing technical difficulties" right now.

piyushk, 2015-07-28:
The wiki's updated. Not sure about the technical difficulty, by the mysql database is administered by the CS admin. The downtime should be temporary.

jack-oquin, 2015-07-29:
In the "Manually Configure Network" section of the "Software installation" page, users are told to get this script: 

*  http://www.cs.utexas.edu/~larg/bwi/scripts/setup_networking.sh

Since networking is not set up yet, how do they copy that script onto the new machine?

jack-oquin, 2015-09-16:
I re-wrote the robot laptop install page to do the initial installation using wired ethernet.

Then it creates the wireless configuration manually:

 * http://www.cs.utexas.edu/~larg/bwi/index.php/Setup_a_robot_laptop_from_scratch#Wireless

That makes it clearer what is actually going on.

@piyushk: please review that my translation looks correct.

closed 2018-01-27 by jack-oquin