utexas-bwi/bwi #32: rosinstall: update lists of source repositories

jack-oquin, 2015-08-06:
Once multi_level_map is up-to-date in the ROS package repositories, it should be deleted from indigo.rosinstall.

@piyushk : What should we do about hydro.rosinstall? It probably won't work anymore.

* Delete it?
* Update it to use the hydro_devel branches? They will be rather down-level, compared to the sources on roberto.

jack-oquin, 2015-08-07:
At this point, I am inclined to just delete hydro.rosinstall and update the README accordingly.

piyushk, 2015-08-07:
Sounds good to me.

Piyush

jack-oquin, 2015-08-07:
OK. I'll wait to make the update until I get back to town. By then the multi_level_map debs should be up-to-date and I can delete that repo from indigo.rosinstall.

closed 2016-07-07 by jack-oquin