utexas-bwi/bwi #36: Verify lab user/group manipulation instructions + Easier password changes.

piyushk, 2015-09-16:
I've updated http://www.cs.utexas.edu/~larg/bwi/index.php/Manipulating_users_and_groups with instructions for manipulating users and group. @joq, can you test these when you get some free time?

To make it easier for changing passwords:
 - [ ] Change password of bwi template to match fri template.
 - [ ] Create and document wrapper around password change script.

jack-oquin, 2015-09-17:
Sounds good. I'll try it tomorrow.

jack-oquin, 2015-09-17:
I wonder if we should just put everyone in the ``bwi`` primary group?

jack-oquin, 2015-09-17:
I cleaned up the PHP LDAP admin sections a little. We probably want to leave that in as an alternative method.

jack-oquin, 2015-09-22:
I tried to create the example user account:
```
sudo /home/bwilab/scripts/bwiuseradd fri abc "ABC DEF" fri
```
The operation appeared to work, but I was unable to login with it using the documented default password.

The instructions do not make it clear that all users *must* be added to group ``dialout``. Adding that user to the group did work, but I was unable to verify it by logging in.

Deleting the user worked, and removed it from ``auto.home``, but not from ``dialout``.

jack-oquin, 2015-09-22:
After some delay, ``abc`` did disappear from the ``dialout`` group. Those ugly time-to-live problems seems to happen even when using the refresh button the phpLDAPadmin screen.

piyushk, 2015-09-24:
I've made the following changes to the scripts:
 1. bwiuseradd automatically adds a user to the group dialout.
 2. Changed bwi and fri template passwords to the documented one.
 3. bwiuserdel now deletes a user from all groups.

These changes should fix aforementioned issues you pointed out.

piyushk, 2015-09-24:
I've also created a convenience wrapper around the password change script and documented it here: http://www.cs.utexas.edu/~larg/bwi/index.php/Updating_Your_Lab_Password

jack-oquin, 2015-09-24:
The scripts work OK now, I was able to create "def", login, verify group membership, and then delete that user.

The user "abc" seems to exist already and has a different password. I assume that's left around from earlier testing.

:+1: LGTM

closed 2015-09-24 by jack-oquin