utexas-bwi/bwi #37: Zapp Lab Machine not able to connect to internet

maxsvetlik, 2015-10-20:
Zapp.csres is not online. The user bwilab is able to log in, but no other users due to the lack of communication with the LDAP server

jack-oquin, 2015-10-20:
I just logged in via ssh with partial success. My id and password were accepted, but my home directory is missing.

jack-oquin, 2015-10-21:
There is something peculiar about the network administration. The machine was configured to use wifi. Is there even a wifi device present?

It is automatically connecting to the ethernet, but not via network manager, which leads me to suspect someone has been messing with the network configuration.

Running this seems to clear the problem, but does not answer the question of how the autofs daemon was stopped:
```
sudo service autofs restart
```

piyushk, 2015-10-21:
All desktops have a wireless interface, but it is not typically used. I
believe current settings allow all users to connect through it. Do you
think it needs to be disabled?

/var/log should contain some entries for when the autofs daemon stopped
working.

jack-oquin, 2015-10-21:
I don't know. The only visible problem was that it tried and failed to connect via wifi. Maybe that had nothing to do with the autofs problem.

jack-oquin, 2015-10-21:
As it seems to work again, I am closing this issue. Please re-open if it recurs.

maxsvetlik, 2015-10-28:
This same issue happened this week on kif. The solution was the same; log in to tty and restart autofs

jack-oquin, 2015-10-28:
I was afraid that might happen. 

Perhaps there is some way to automatically respawn the daemon.

jack-oquin, 2015-10-28:
Possibly relevant links:

* https://bugs.launchpad.net/ubuntu/+source/autofs5/+bug/1007273
* http://askubuntu.com/questions/302514/upstart-job-fails-to-respawn-and-hangs-when-tried-to-stop-restart-it

jack-oquin, 2017-12-16:
We had a similar problem with Ubuntu 16.04 on nibbler yesterday, and @rdelfin ended up reinstalling it.