utexas-bwi/bwi #41: compilation issue in segway_rmp package

jsinapov, 2016-07-08:
Upon setting up a workspace, the first catkin_make run produces the following error:

/home/users/jsinapov/catkin_ws/src/segway_rmp/src/segway_rmp_node.cpp:32:44: fatal error: segway_rmp/SegwayStatusStamped.h: No such file or directory
 #include "segway_rmp/SegwayStatusStamped.h"
                                            ^
compilation terminated.

Chances are, the CMakeLists.txt needs to specify that the message needs to be compiled before the segway_rmp_node is compiled.

jack-oquin, 2016-07-08:
Does running it once or twice again resolve the error? If so, that's almost certainly the problem.

But, it looks like this issue is probably in the [segway_rmp](https://github.com/segwayrmp/segway_rmp) repository.

jack-oquin, 2017-12-16:
I believe this is fixed now.

closed 2017-12-16 by jack-oquin