utexas-bwi/bwi #49: Desktop installation of BWI

nitin5, 2017-09-11:
Greetings 
I am trying to install BWI code on my desktop. I do not have access to Segbot and will like to use gazebo simulations. I am interested in Multi_level_map navigation package. 
I am getting following errors in following listed instructions

 **$rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bwi_virtour: Cannot locate rosdep definition for [web_video_server]
segbot_navigation: Cannot locate rosdep definition for [eband_local_planner]
segbot_sensors: Cannot locate rosdep definition for [hokuyo_node]
segway_navigation_apps: Cannot locate rosdep definition for [frontier_exploration]
segbot_bringup: Cannot locate rosdep definition for [segway_rmp]**

On doing catkin build, following error is reported
**Errors     << velodyne_driver:make /home/nitin/catkinws_bwi/logs/velodyne_driver/build.make.000.log                               
In file included from /home/nitin/catkinws_bwi/src/velodyne/velodyne_driver/src/lib/input.cc:34:0:
/home/nitin/catkinws_bwi/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:37:18: fatal error: pcap.h: No such file or directory
compilation terminated.
make[2]: *** [src/lib/CMakeFiles/velodyne_input.dir/input.cc.o] Error 1
make[1]: *** [src/lib/CMakeFiles/velodyne_input.dir/all] Error 2
make: *** [all] Error 2
cd /home/nitin/catkinws_bwi/build/velodyne_driver; catkin build --get-env velodyne_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................................
Failed     << velodyne_driver:make                           [ Exited with code 2 ]                                               
Failed    <<< velodyne_driver                                [ 6.1 seconds ]                                                      
Abandoned <<< bwi_launch                                     [ Unrelated job failed ]                                             
Abandoned <<< bwi_kr_execution                               [ Unrelated job failed ]                                             
Abandoned <<< bwi_rqt_plugins                                [ Unrelated job failed ]                                             
Abandoned <<< bwi_scavenger                                  [ Unrelated job failed ]                                             
Abandoned <<< bwi_services                                   [ Unrelated job failed ]                                             
Abandoned <<< bwi_tasks                                      [ Unrelated job failed ]                                             
Abandoned <<< bwi_virtour                                    [ Unrelated job failed ]                                             
Abandoned <<< segbot_bringup                                 [ Unrelated job failed ]                                             
Abandoned <<< segbot_gazebo                                  [ Unrelated job failed ]                                             
Abandoned <<< segbot_gui                                     [ Unrelated job failed ]                                             
Abandoned <<< segbot_led                                     [ Unrelated job failed ]                                             
Abandoned <<< segbot_logical_translator                      [ Unrelated job failed ]                                             
Abandoned <<< segbot_navigation                              [ Unrelated job failed ]                                             
Abandoned <<< segbot_sensors                                 [ Unrelated job failed ]                                             
Abandoned <<< segbot_simulation_apps                         [ Unrelated job failed ]                                             
Abandoned <<< stop_base                                      [ Unrelated job failed ]                                             
Abandoned <<< velodyne_pointcloud                            [ Unrelated job failed ]                                             
Finished  <<< utexas_gdc                                     [ 2.9 seconds ]                                                      
Finished  <<< bwi_msgs                                       [ 14.6 seconds ]                                                     
[build] Summary: 23 of 41 packages succeeded.                                                                                     
[build]   Ignored:   7 packages were skipped or are blacklisted.                                                                  
[build]   Warnings:  None.                                                                                                        
[build]   Abandoned: 17 packages were abandoned.                                                                                  
[build]   Failed:    1 packages failed.                                                                                           
[build] Runtime: 1 minute and 28.9 seconds total.                                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.**

I understand that all the dependencies have not be installed by rosdep because of which further compilation errors are being reported. Is there a minimal subset for multi-level-map navigation and gazebo documented which i can try?

Rgds
Nitin

jack-oquin, 2017-09-12:
What ROS version are you running? On what Ubuntu operating system?

nitin5, 2017-09-12:
I am using ROS Kinetic on Ubuntu 16.04.

rdelfin, 2017-09-14:
Our codebase only supports Ubuntu 14.04 with ROS Indigo (check issue #44). We recommend you downgrade your version of Ubuntu in the meantime while this is resolved.

jack-oquin, 2017-09-19:
see also: utexas-bwi/bwi_common#95 and utexas-bwi/segbot#77

jack-oquin, 2017-10-23:
Closing this as it is essentially a duplicate of the others linked above.

closed 2017-10-23 by jack-oquin