utexas-bwi/bwi #52: Bring segbot_arm packages into the main repositories

nickswalker, 2018-01-26:
Bringing these packages in has a few benefits:

* Simplifies the usage of the arm robot. No special workspace-creation procedures
* Sharing previously duplicated code will make future migrations, bug fixes easier
* Better integration will make moving the arm to an updated RMP base easier in the future

The cons include more packages in a default BWI workspace, but as @jack-oquin has highlighted, this is something that can be handled with `catkin_tools` blacklists. This will also have the effect of moving many previously centralized bits of arm infrastructure to their own separate corners.

After some discussion, @jack-oquin and I agreed that we're in a good position to do this now, as we're already going to induce a lot of changes with the Kinetic migration. I am especially interested in doing this quickly as @harely22 and @thomason-jesse are setting up to do more work on the platform.

I have already merged the launch/description/bringup components into the appropriate packages on the Kinetic branch, and I'll be moving the remaining packages soon.

jack-oquin, 2018-01-26:
As Nick said, this seems like a good idea to me.

jack-oquin, 2018-01-26:
I'd like to get our Kinetic branches in good shape to share **bwi_common** with the RoboCup@Home team.

shihyunlo, 2018-01-26:
Would segbot_arm launch file remain separate from segbot_v3/v4? If so, it
sounds great to me; if not, we might need to test out the performance of
regular tasks (such as visit_door_list), to make sure both navigation and
kr_execution are not much slowed down.

nickswalker, 2018-01-26:
The launch files will be in the same package now (bwi_launch), but they will remain separate. There should be no change at run time for any non-armed robot.

shihyunlo, 2018-01-26:
Sounds great, I am happy with it.

nickswalker, 2018-01-26:
This is done.

segbot_arm_perception was easy to promote into bwi_common (nice win for RoboCup), but segbot_arm_tasks is stuck separate from bwi_tasks because of segbot_arm_manipulation. We need a better abstraction over basic manipulation capabilities.

jack-oquin, 2018-01-26:
Our current launch scripts have evolved into a bit of a hairball over the years. Fixing that is clearly outside the scope of this issue.

I have some ideas how to make them much more modular, and may open a separate issue for that.

jack-oquin, 2018-01-26:
Note that due to new packages and dependencies, people will probably want to check out new workspaces from scratch.

jack-oquin, 2018-01-26:
I am getting an unresolved dependency:
```
$ rosdep install --from-paths src --ignore-src --rosdistro=kinetic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
kinova_arm_moveit_demo: Cannot locate rosdep definition for [pr2_moveit_plugins]
```

jack-oquin, 2018-01-26:
That problem is in our copy of the **kinova-ros** repo. Putting CATKIN_IGNORE in the **kinova_arm_moveit_demo** package there works around it temporarily.

nickswalker, 2018-01-26:
It seems that there isn't a kinetic release of that package yet. I don't know that we have every referenced that demo, so although I am somewhat uncomfortable with the practice, I think it may be fine to commit the ignore file and let the kinova folks fix it whenever they get around to it.

jack-oquin, 2018-01-26:
I cherry-picked 59e78f16bd06e7197b9419e2d31fd92d963d5163 into master, because the generic BWI install instructions use that branch.

jack-oquin, 2018-01-26:
I agree about the CATKIN_IGNORE hack, but went ahead to avoid messing up other people's work flow.

jack-oquin, 2018-01-26:
WIth the CATKIN_IGNORE I can download, build and run some basic BWI navigation tests.

closed 2018-01-26 by jack-oquin

jack-oquin, 2018-01-26:
xref: utexas-bwi/kinova-ros#3